MaJerle/lwcell #4: size of compiled binary

danyhm, 2018-12-19:
Hello again,

I just tried to compile the library and integrate it into my project but the linker complained about not having enough size which is weird because I've a STM32F091RCT and it has 256Kb of flash. from the source files that i looked , it doesn't sound that much to me. Do you have any estimation of the compiled library ?

MaJerle, 2018-12-19:
Hello @danyhm 

I assume your implementation does not suffer due to flash (except if your application has already 240kB usage) but from RAM. Please check compiler (linker) output where it fails and which memory is missing.

I did not yet check for code size.

danyhm, 2018-12-19:
@MaJerle 
No my flash is almost empty (around 30kb for RTOS and a few simple functions.)
I don't think it's RAM , i have 32kb and RTOS is using just 10kb.
the linker says `Main\Main.axf: Error: L6407E: Sections of aggregate size 0x134f0 bytes could not fit into .ANY selector(s).`
and that's 79,088 bytes not being fitted. I think It's flash

danyhm, 2018-12-19:
I Take that back, you were right it's ram. i'm going to investigate what is exactly using that much ram

MaJerle, 2018-12-19:
Maybe you are victim of direct copy/paste LL driver for STM32F0.
Look size of memory I used in case of WIN32. This memory is used for internal dynamic allocation for every command and received data packet. It is used by `gsm_mem` layer.

https://github.com/MaJerle/GSM_AT_Lib/blob/master/src/system/gsm_ll_win32.c#L189

danyhm, 2018-12-19:
it was 
`static uint8_t memory[0x10000];             /* Create memory for dynamic allocations with specific size */`
in gsm_ll_init function. 
i forgot that library uses it's own memory management system. just like stemwin

closed 2018-12-19 by MaJerle