MaJerle/lwcell #5: sim7000 auto baudrate

Tasosgi, 2019-01-30:
The sim7000 from default is on auto baud rate. The MCU must transmit some AT after sim7000 reset  for sync. Is there a provision for this?

MaJerle, 2019-01-30:
Library doesn't care about any low-level stuff, like GPIO/UART initialization for drivers. That's part of driver implementation.

Tasosgi, 2019-01-30:
I think the gsm_reset_with_delay() is part of the library. I would expect some AT there for sync before the first command to GSM. I am surprising how the examples are working without these. Maybe only my version of SIM7000 is with auto baudrate enabled as default.

Also congratulations for your work and thank you for sharing it with us!

alxkbr, 2020-05-28:
The same problem with SIM800C