MaJerle/lwcell #6: buffer processing issue

danyhm, 2019-02-04:
There's an issue with the way GSM received data buffer is processed.
1- the function `gsmi_process()` is called to process the buffer.
https://github.com/MaJerle/GSM_AT_Lib/blob/6653f46d22d2c6fcf61003346616afda28fbed78/src/gsm/gsm_int.c#L964
2- in that function after some process the gsmi_parse_received() function is called which handles the events and calls user callback functions if needed.
https://github.com/MaJerle/GSM_AT_Lib/blob/6653f46d22d2c6fcf61003346616afda28fbed78/src/gsm/gsm_int.c#L1131
3- inside that function any user callback function can be called for example for an SMS this line is called 
https://github.com/MaJerle/GSM_AT_Lib/blob/6653f46d22d2c6fcf61003346616afda28fbed78/src/gsm/gsm_int.c#L719

now let's imagine user callback function is called inside gsmi_parse_cmti(). if the user has some blocking code inside his function the whole thread of execution is blocked because of that and if any more data from GSM is received in a considerable amount of time, they will be lost because the processing thread is halted due to user function being the blocking factor here.

this issue is really application dependent. in a house security system that I'm designing this can be a fatal point but I'm still working on it. anyway i still find your library inspiring and I'm writing my own library using your ideas.

a work around for this that i can think of is to make the processing thread completely non blocking by using OS queues. the user thread has to process this messages pushed by processing thread.

MaJerle, 2019-02-04:
This is not considered an issue. There is no point of doing blocking tasks in callback. If you need blocking tasks, write information to queue and process in another thread.

danyhm, 2019-02-04:
that can't be used for results not generated from API calls. for example a received SMS is only handled as callback.

MaJerle, 2019-02-04:
@danyhm yes, you get notification that SMS has been received. Now you cannot read it as blocking mode from callback. If you need to read it blocking mode, create new thread for SMS read and for every received notification save your SMS memory&number and then in separate thread read that SMS blocking mode.

That's how AT commands work in general.

danyhm, 2019-02-04:
@MaJerle yes i know , that is one way to handle it. but creating more threads always consume ram which is not suitable for ram constrained applications. i haven't come up with a better solution myself yet.

MaJerle, 2019-02-04:
> @MaJerle yes i know , that is one way to handle it. but creating more threads always consume ram which is not suitable for ram constrained applications. i haven't come up with a better solution myself yet.

Of course there is an option to use separate threads, but then you need to push every possible callback to a queue and use separate thread to read callbacks and send them to user callbacks. Point is that you cannot block producer nor processing threads otherwise you will end up in dead-lock.

closed 2019-02-13 by danyhm