MaJerle/lwcell #8: Getting gsmERRMEM error after some usage or overload

oguzhancakan, 2019-04-19:
Hi Mr. Tilen, 

First of all thanks for this lib. I am using this lib for mqtt conn but after a while or when I send a lot of package from server to the topic which the board uses this library subscribed, it generates memory error. I tried to trace code and solve the problem but I couldn't.

How can I free the memory, or just restart all the library from beginning by freeing the memory. I had a solution by using two stm32f4 mcus, for one this library, another for others tasks. When this error occured , I can restart the mcu and the other mcu can continue to its tasks. But I know, it is not the appropriated way.

(By the way, the code I used is your mqtt client api demo.)
So I need your help. Thanks in advance.

Best Regards,
Oguz.

MaJerle, 2019-04-19:
Definitely restarting MCU is bad idea. We need to identify and fix the issue. About MQTT. I'm using the same MQTT API in my ESP8266 and GSM library, code exactly the same, application part the same.

Now what I see optional problem could be that in case of GSM library I do not handle error on data send in all possible cases as they are unknown. It would be good to see from your side if you are receiving some kind of error when AT+CIPSEND is executed.

oguzhancakan, 2019-04-19:
I just tried with your code only, It seems ok. I think my problem is about touchgfx (Problem occurs when touchgfx active). And It brings me again using dual mcu. Can you suggest anything else? Can I run this library from external sdram? (I use stm32f429i disco and it has external sdram) .

Thanks in advance,

Best Regards,
Oguz.

MaJerle, 2019-04-19:
Library does not really care from where you execute it. Also, you can use external memory for library RAM. You need to set ram in LL function for init

oguzhancakan, 2019-04-19:
It maybe hard to achieve for me but I will try. Appreciate for your help. 

Best Regards,
Oguz.

MaJerle, 2019-04-19:
Are you able to use external RAM in your application already? If not, please check examples in STM32CubeF4 package for your board. If yes, I will prepare a very short explanation how to do it.

oguzhancakan, 2019-04-20:
Mr. Tilen,

You are really good man, appreciate a lot.

Yes, I am able to use external RAM but touchgfx part is using it. Actually I didn't manage it, but with your help, I can run it on external ram. The only thing I have to find out which address space is used by touchgfx. 

Best Regards,
Oguz.

closed 2019-07-22 by MaJerle