MaJerle/lwcell #9: MQTT Reconnection Issue

oguzhancakan, 2019-04-22:
Hello Mr. Tilen,

I tried to figure out what happens if SIM800 power downs(mcu and sim800 power source are different) for a while or network corrupts when I use mqtt api. I experienced that mqtt api blocks itself at publish call. (For example at line 95 here /snippets/mqtt_client_api.c)

After some trace, I realized that at apps/mqtt/gsm_mqtt_client_api.c Line: 444
`    if (gsm_mqtt_client_publish(client->mc, topic, data, GSM_U16(btw), qos, retain, NULL) == gsmOK) {
        gsm_sys_sem_wait(&client->sync_sem, 0);
        res = client->sub_pub_resp;
    }`

Here `gsm_sys_sem_wait(&client->sync_sem, 0);` blocks the program. And I changed it by adding timeout to:

`if (gsm_mqtt_client_publish(client->mc, topic, data, GSM_U16(btw), qos, retain, NULL) == gsmOK) {
       if(gsm_sys_sem_wait(&client->sync_sem, 10000) == GSM_SYS_TIMEOUT){
        	res = gsmERR;
        }else{
        	res = client->sub_pub_resp;
        }
    }`

After this in mqtt_client_api.c I am checking return value of publish. If it is gsmERR I am restarting SIM800, reattach to network and reconnect mqtt sequence. 

Actually I couldn't find any other way to detect corruption and solution for blocking.What do you think about this solution? Is it ok or do you suggest more elegant way?

Best Regards,
Oguz.

MaJerle, 2019-04-22:
This is in fact not the best way as device shall report and detect "problems" on lower part, not application part. Semaphore shall get unlocked in connection callback function, which is never called as there is no feedback from device. It may be a bug in the library related to how is connection handled. This would need to be checked against your case.

oguzhancakan, 2019-04-22:
For may case It seems that solved my problem. But usually I got timeout error and I handle it at application part. It comes with some natural latency due to restart SIM800 and connection but it is acceptable. Maybe you can consider this as a bug report. I will check updates. If you solve this problem please let me know.

Best Regards,
Oguz.

vinhemotiv, 2019-07-23:
I got the same issue. The whole application was blocked and crashed.

MaJerle, 2019-07-24:
> I got the same issue. The whole application was blocked and crashed.

Can you share log? I left it during the night active with cayenne MQTT server and it still works using MQTT API.

MaJerle, 2019-07-24:
I think I found potential issue.
`AT+CIPSEND` command returned `ERROR`. Because of that, stack wants to manually close the connection, but `AT+CIPCLOSE` also returned `ERROR`, hence `CLOSED` event was never received and semaphore not unlocked.

vinhemotiv, 2019-07-27:
while (1) {
        /* Make a connection */
        printf("Joining MQTT server\r\n");

        /* Try to join */
        conn_status = gsm_mqtt_client_api_connect(client, "222.255.11.94", 1883, &mqtt_client_info);
        if (conn_status == GSM_MQTT_CONN_STATUS_ACCEPTED) {
            printf("Connected and accepted!\r\n");
            printf("Client is ready to subscribe and publish to new messages\r\n");
        } else {
            printf("Connect API response: %d\r\n", (int)conn_status);
            gsm_delay(5000);
            continue;
        }
        while(1)
	  {
		osDelay(500);
	  }
}
I have a above thread. I just call gsm_mqtt_client_api_connect and then do nothing.
The whole application(free rtos) will be hang up after a while. If I remove out the gsm_mqtt_client_api_connect, the application works well.
Not sure why? :(

MaJerle, 2019-07-27:
No.it is not because of MQTT api. If connect function goes through then it is something else. Mine is running well for past 2 days.

Maybe stack for thread.

vinhemotiv, 2019-07-28:
Connect function goes through okay. This is my code to create a stack of gsm thread as example does:
static void
init_thread(void const* arg) {
    printf("Starting GSM application!\r\n");
    /* Initialize GSM with default callback function */
    if (gsm_init(gsm_callback_func, 1) != gsmOK) {
        printf("Cannot initialize GSM-AT Library\r\n");
//        while (1) {
//					HAL_GPIO_TogglePin(GPIOA, GPIO_PIN_14);
//					osDelay(500);
//				}
    }

    /* Configure device by unlocking SIM card */
    if (configure_sim_card()) {
        printf("SIM card configured. Adding delay to stabilize SIM card.\r\n");
        gsm_delay(10000);
    } else {
        printf("Cannot configure SIM card! Is it inserted, pin valid and not under PUK? Closing down...\r\n");
        gsm_delay(1000);
    }

    /* First enable SMS functionality */
    if (gsm_sms_enable(NULL, NULL, 1) == gsmOK) {
        printf("SMS enabled. Send new SMS from your phone to device.\r\n");
    } else {
        printf("Cannot enable SMS functionality!\r\n");
        //while (1) { gsm_delay(1000); }
    }

    /* Now send SMS from phone to device */
    printf("Start by sending SMS message to device...\r\n");

    while (1) {
        /* Connect to network for TCP/IP */
        if (gsm_network_attach(NETWORK_APN, NETWORK_APN_USER, NETWORK_APN_PASS, NULL, NULL, 1) == gsmOK) {
            printf("Attached to network!\r\n");

            /* Start netconn thread */
            gsm_sys_thread_create(NULL, "netconn_thread", (gsm_sys_thread_t)mqtt_client_api_thread, NULL, GSM_SYS_THREAD_SS, GSM_SYS_THREAD_PRIO);
            break;
        } else {
            printf("Cannot attach to network!\r\n");
            gsm_delay(1000);
        }
    }

    while (1) {
			HAL_GPIO_TogglePin(GPIOA, GPIO_PIN_14);
      gsm_delay(1500);
    }
    
    osThreadTerminate(NULL);                    /* Terminate current thread */
}

vinhemotiv, 2019-07-28:
Can you share the main file? application code to use gsm module? I'm not sure what I'm wrong.

MaJerle, 2019-07-28:
Download resources repository and run example there. Or even better, take visual studio examples. Use VSCommunity if you need free edition.

vinhemotiv, 2019-07-28:
[FreeRTOSConfig.h.txt](https://github.com/MaJerle/GSM_AT_Lib/files/3439263/FreeRTOSConfig.h.txt)
Hi MaJerle,
This is my FreeRTOS config.h. I refer your freertos config.h file but I can't comment out the 
#define xPortSysTickHandler SysTick_Handler
It won't work for me. Could you help to check what's is going wrong with my config file?
Tks alot
I will try to debug the overflow stack

closed 2019-07-28 by MaJerle