MaJerle/lwcell #10: [Question] Any plan to CMake support?

tan-wei, 2019-04-28:
Thanks for this great project. It seems that there're seldom AT command open source projects I can find.
As a cross-platform project, is there any plan that provide CMake support? Thanks very much!

MaJerle, 2019-04-28:
When you say CMake support, what is your request? Is source files not enough? Thanks for clarifying.

tan-wei, 2019-04-28:
Thanks for replying.
After some hours reading source codes and documents, it seems that the project mainly used on embedded system devices, then the other systems (`Windows`, `Linux` and `MAC`). So I think CMake is not that must for systems which can not cross-compiled in `Windows`, `Linux` or`MAC`.
Some platform specific functions should be implemented by users, which can not be done by CMake automatically, it makes CMake useless in these specific system. When I first ask this question, I don't realize the project is highly depending on the platform. Sorry for that.
However, it may convinient the users who want to use the project as a `library` in `Windows`, `Linux` or`MAC`, CMake will make it quite easy to generate the corresoponding library. Actually, it is not must for this project. If necessary, I would fork the project and write a CMake file. Maybe someone like me want that. Just maybe...
I will try the project under windows soon. Thanks very much! Hope I do not waste of your time.
Hope you have a nice weekend!

tan-wei, 2019-04-28:
So is the issue going to be closed?

MaJerle, 2019-04-28:
Library consists of 2 parts, one is platform independent and second is highly-platform dependent. Second part includes special functions for operating system and communication between host (Windows, etc, embedded MCU, etc) and ESP8266 and these must be developed specific for each system.

I will close the issue now.

closed 2019-04-28 by MaJerle