MaJerle/lwcell #12: Bug in gsmi_send_msg_to_producer_mbox()

dslc, 2019-05-31:
In `gsmi_send_msg_to_producer_mbox()` ...

If the library fails to place the message in the producer queue - and subsequently deallocates the message memory and resets the pointer as it does at
https://github.com/MaJerle/GSM_AT_Lib/blob/86929f299f7b2546af4d258f214e9629ef6322bb/src/gsm/gsm_int.c#L2114 - then it should return straight away. Otherwise it will attempt to dereference a null pointer.

Perhaps the simplest solution is to replace https://github.com/MaJerle/GSM_AT_Lib/blob/86929f299f7b2546af4d258f214e9629ef6322bb/src/gsm/gsm_int.c#L2115 with

```
return gsmERR;
```

MaJerle, 2019-05-31:
Thanks for reporting. i agree that the problem is one line lower.

https://github.com/MaJerle/GSM_AT_Lib/blob/86929f299f7b2546af4d258f214e9629ef6322bb/src/gsm/gsm_int.c#L2118

If statement should check res first, then msg.

dslc, 2019-06-01:
Thanks for the response. I see you have already committed the bug fix. I will close this issue now.

closed 2019-06-01 by dslc