MaJerle/lwcell #13: MQTT disconnection on every single receive

oguzhancakan, 2019-06-22:
Hi Mr. Tilen,

I am using and testing this library for real product and almost everything fine.

But for example, when I use exactly same code with: GSM_AT_Lib_res/docs/examples/_example_mqtt_client.c mqtt 

MQTT works fine except I publish something to topic which this library subscribed. It takes the message but after that it disconnects. At every single received message, it disconnects and reconnects. (retriess++ in the code) 

Actually this may be acceptable because of the latency is a few seconds to reconnect. But if the message's retain flag was set which sent to the device, it takes message, then disconnects, then reconnects, and again takes the message.. The same loop. This case makes the device unable to publish any message. 

Do you have any suggestion to solve this problem? 

Thanks in advance,

Oguz.

MaJerle, 2019-06-22:
Best would be if you couod send AT log.

oguzhancakan, 2019-06-24:
You are absolutely right. I was using HiveMQ public server then I changed it to my custom mosquitto server then problem solved. 

Appreciate again for your help.

closed 2019-06-24 by MaJerle