MaJerle/lwcell #14: MQTT with SSL

ProfEng, 2019-06-28:
Hi Mr. Tilen,

Did you implement any MQTT example with SSL encryption. I changed:

`res = gsm_conn_start(&client->conn, GSM_CONN_TYPE_TCP, host, port, client, mqtt_conn_cb, 0);`

with: 
`res = gsm_conn_start(&client->conn, GSM_CONN_TYPE_SLL, host, port, client, mqtt_conn_cb, 0);`

at gsm_mqtt_client.c Line: #1103 and changed port of broker to 8883. It gives me GSM_MQTT_CONN_STATUS_TCP_FAILED error code. 

Do I miss anything? 

Appreciate for your help.

Aykut.

oguzhancakan, 2019-06-28:
Not exactly sure but found this:

[](https://stackoverflow.com/questions/38713176/iot-cant-connect-to-the-server-with-mqtttls)

What do you think?

MaJerle, 2019-06-30:
I find it very hard to use SSL with SIM800. There should be a command AT+CIPSSL to enable SSL but seems that it doesn't work always. So technically SSL shall be possible, somehow though.

gxnt-samir, 2021-03-13:
Hello @MaJerle 

I am curently using your lwgsm and i really appriciate your effort of crafting such a beautiful and well organised library.

I have some questions while going through your sample example of mqtt and if you can help me understanding it little better it will be great.

Background : I saw you are using default SSL certificate of SIM800 TCP AT cmd in MQTT communication. when i enqury with SIMCOM, they told me that the module actually donot have any default certificate for MQTT brokers. To use the SSL over MQTT we need to push/write the rootca, client certificate and key to the SIM800 filesystem.

Query:
1. Do you have any plans to add those functionality to your library? If you need help with AT cmds i can get that for you.
2. If your plan is to do it later, its fine i can help improve and contribute that portion. But i am currently having trouble to understand followings
-> If i have already to started a HTTPs connection using client_connect API which internally calls lwgsm_conn_start API, and after pushing the new certificate to SIMCOM filesystem, i only need to use the SSL for lwgsm_mqtt_client_connect API. Is it possible? 
-> Do i have to do an reinit of library or i can start using the SSL from middle of the application. This question is not hardware dependent but to know your current library architecture support or not.
-> Also in your documentation you have mentioned using SSL only 1 simultaneous connection is possible. is it a limitation of the library or hardware.

Note: I also found out from SIMCOM that the newer virsion(released end of dec 2020) of SIM800 series have larger filesystem memory(32MB) to store SSL and support SSL internally. I am not sue but from your previous comment i saw you tried the SSL and didnot work. But may be the older version of the chip didnot support SSL. Just my guess.

I am looking forward for your reply.

Thanks 
Samir

MaJerle, 2021-03-19:
So far I have no defined roadmap of the lib. I'm more than open to get proposals from community.