MaJerle/lwcell #15: ..\GSM_AT\gsm\gsm_threads.c(171): error:  #167: argument of type "gsm_msg_t **" is incompatible with parameter of type "void **"

vinhemotiv, 2019-07-21:
When I build with MDK5-ARM using Keil. I got the the following errors
..\GSM_AT\gsm\gsm_threads.c(171): error:  #167: argument of type "gsm_msg_t **" is incompatible with parameter of type "void **"

Can you try to help?

MaJerle, 2019-07-21:
https://github.com/MaJerle/GSM_AT_Lib/blob/4fa85e1c1324e91fd63b0e8dc250d60d4c0f439e/src/gsm/gsm_threads.c#L167

Nothing there to trigger an error.

vinhemotiv, 2019-07-21:
Line 171.

MDK5 can’t build it with Keil

vinhemotiv, 2019-07-21:
..\GSM_AT\gsm\gsm_timeout.c(83): argument of type "gsm_timeout_t **" is incompatible with parameter of type "void **"
https://github.com/MaJerle/GSM_AT_Lib/blob/4fa85e1c1324e91fd63b0e8dc250d60d4c0f439e/src/gsm/gsm_threads.c#L171

I got a lot of the incompatibility error when building them with MDK5

MaJerle, 2019-07-21:
Try to pull changes and recompile.

vinhemotiv, 2019-07-22:
That works. Thank you alot!

closed 2019-07-22 by vinhemotiv