MaJerle/lwcell #16: Can't subcribe a topic with mqtt client on Sim800

vinhemotiv, 2019-07-22:
Hi,
I'm testing mqtt client on sim800 and I run the code in example
Look like I can't subscribe a topic even the server was GSM_MQTT_CONN_STATUS_ACCEPTED:

            if (status == GSM_MQTT_CONN_STATUS_ACCEPTED) {
                printf("MQTT accepted!\r\n");
                /*
                 * Once we are accepted by server, 
                 * it is time to subscribe to different topics
                 * We will subscrive to "mqtt_gsm_example_topic" topic,
                 * and will also set the same name as subscribe argument for callback later
                 */
                gsm_mqtt_client_subscribe(client, "gsm_mqtt_topic", GSM_MQTT_QOS_EXACTLY_ONCE, "gsm_mqtt_topic");
                
                /* Start timeout timer after 5000ms and call mqtt_timeout_cb function */
                gsm_timeout_add(5000, mqtt_timeout_cb, client);
            } else {
                printf("MQTT server connection was not successful: %d\r\n", (int)status);
                
                /* Try to connect all over again */
                example_do_connect(client);
            }

MaJerle, 2019-07-22:
How do you know you cannot subscribe? is connection closed? Is it closed due to remote side?

vinhemotiv, 2019-07-22:
I'm testing with 
gsm_mqtt_client_connect(mqtt_client, "test.mosquitto.org", 1883, mqtt_cb, &mqtt_client_info);

Here is the serial log:

Application running on ..!
                                                    
Starting GSM application!
                                                     
Library initialized!
                                                          
SIM card configured. Adding delay to stabilize SIM card.
                      
SIM device SMS service is ready!
                                              
Network registration status changed. New status is: Connected to home network!

Operator long name: Viettel Mobile
                                            
Network operator RSSI: -51 dBm
                                                
SMS enabled. Send new SMS from your phone to device.
                          
Start by sending SMS message to device...
                                     
Attached to network!
                                                          
MQTT Client ID: 0123456789abcdef
                                              
MQTT accepted!
                                                                
Publishing 10...
                                                              
MQTT client disconnected!
                                                     
Publish event, user argument on message was: 10
                               
MQTT client disconnected!

MaJerle, 2019-07-22:
Looks like it is closed due to remote side and not by library.

vinhemotiv, 2019-07-22:
I can use the TPC connection in the netconn_thread, TCP is connected and it works well.

But when using a MQTT client connect, it failed and time out.

Not sure why

closed 2019-07-22 by vinhemotiv

MaJerle, 2019-07-22:
You found the issue related to non-netconn approach? Or why did you close it?

vinhemotiv, 2019-07-22:
Yes. It was my fault. There is a problem with my simcard :(
Thank you a lot on your support. :)
Will contact you on other projects :)

MaJerle, 2019-07-30:
Issue finally related to customer's imolementation of LL druver. DMA interrupt was not properly cleared causing endless interrupt jumps on HT event.

vinhemotiv, 2019-07-31:
Yep. Thank you a lot for your help!