MaJerle/lwcell #19: SIM ready call gsmi_get_sim_info(0) cause system deadlock

wishmiss, 2019-12-26:
1. 

        /*
         * In case SIM is ready,
         * start with basic info about SIM
         */
        if (gsm.m.sim.state == GSM_SIM_STATE_READY) {
            gsmi_get_sim_info(0);
        }

**#define GSM_CFG_INPUT_USE_PROCESS 1**

now is in gsm_input_process and gsm_core_lock,  gsmi_get_sim_info(0) call gsmi_send_msg_to_producer_mbox will be failed

2. how to deal with **GSM_CMD_SIM_PROCESS_BASIC_CMDS** command?

3. gsmr_t gsmi_get_sim_info(const uint32_t blocking) put in gsm_sim.c (gsm_sim_get_info(const uint32_t blocking) will be better

MaJerle, 2019-12-26:
1. Did you try it or is it from code review only?

wishmiss, 2019-12-26:
I want to port this library to linux platform and use pthread_mutex_lock.
Calling "configure_sim_card()", it can't return. So I traced the code.

MaJerle, 2019-12-26:
Mutex must support recursive lock. Then it will work well.

wishmiss, 2019-12-26:
OK. I will double check my code again. Thank you!

wishmiss, 2019-12-26:
Set the mutex attribute **PTHREAD_MUTEX_RECURSIVE**,  It work!

closed 2019-12-26 by wishmiss