MaJerle/lwcell #20: dtmf

atulgo, 2019-12-30:
Detection of DTMF tones - SIM800L module.

I get "OK" when I send 'AT+DDET=1,1000,1,1* ' over uart .
But it doesnt seem to detect anything afterwards.
Perhaps I am doing something wrong.
Would appreciate some feedback.

regards
Atul
ref : *SIM800 Series_AT Command Manual_V1.09

MaJerle, 2020-01-02:
What do you expect to get? OK seems correct answer to this command.

atulgo, 2020-01-03:
ya ...OK is fine. But its afterwards that i face problem.
Because DTMF is not being detected. 
This is what happens : 
1. Put sim 'A' in modem.
2. Connect to modem over serial using Minicom.
3. Turn on DTMF detection.
4. Initiate a call to modem using a mobile phone.
5. Dial the digit (1,2.... etc) from the phone.
6. Nothing appears on serial port