MaJerle/lwcell #21: Network Attachment

yldzmuhammed, 2020-01-01:
Hello Tijen,

Fİrstly great thanks for this great job. Congurats!

Secondly, i am trying to make some changes for Quectel M66 module.

I have a question for you.

When i looked at SIM800+Series_TCPIP_Application+Note_V1.01.pdf document, i see that gprs attachment and tcp connection flow (page 7). But different from yours.

In your code (gsm_int.c after line 1523) 

- at+cgact=0
- at+cgact=1
- at+cgatt=0
- at+cgatt=1

and goes on. I dont think your code matches with document. But more importantly how can i implement this for M66?

wishmiss, 2020-01-02:
@yldzmuhammed 

Firstly, before any TCP/UDP related operation is set up, the module should be connected to GSM or GPRS network.

How to make sure the module connected to GSM or GPRS network? You can read SIM800 Series_AT Command Manual_V1.09.pdf chapter 7 and chapter8. 

Before attach the service, if you don't know  the status of the service, detach first.

yldzmuhammed, 2020-01-02:
Dear @wishmiss 

Well, infact I followed @MaJerle 's example project. 

In the example project some codes do the init operations. It checks (gsm_init) CREG but i did not see any CGREG check. 

But the problem is code that before GPRS attachment and initing apn, user name and password tries to activate the context (AT+CGACT=1). 

I am not sure about SIM800 but on M66, you can not do that! Before activation, you should set access point name and other things.

So i am saying that the problem is the operation order. Am i wrong?

If you check the file and lines that i mentioned, you will see it.

Please let me know if i am wrong.

wishmiss, 2020-01-02:
@yldzmuhammed 

So your problem is you didn't know what is the operation order?

**gsm_init-->gsm_reset_with_delay (#define GSM_CFG_RESET_ON_INIT  1  reset order)-->gsmi_parse_cpin-->gsmi_get_sim_info (sim process order)**

In the sim process order, you can see AT+CREG.

excamples->win32->netconn_client_rtos, after **gsm_init,** calling **gsm_network_attach**  to do network attach operation order.

yldzmuhammed, 2020-01-02:
@wishmiss you get me all wrong. But no problem. I have solved my problem. Thanks anyway.

closed 2020-01-02 by yldzmuhammed

MaJerle, 2020-01-02:
@yldzmuhammed do not close this that fast. I can understand your request. AT commands were developed for SIM800 in mind, not Quectel modems. Some modifications might be necessary to support both, but not both have the same at commands.