MaJerle/lwcell #22: gsmi_parse_string return value contains CRLF

wishmiss, 2020-01-02:
calling gsmi_parse_string will return a new string end of CRLF, Is this result expected?

In my project, i combining model serial number with other string, subsequent processing errors due to this extra string.

`uint8_t
gsmi_parse_string(const char** src, char* dst, size_t dst_len, uint8_t trim) {
    const char* p = *src;
    size_t i;

    if (*p == ',') {
        ++p;
    }
    if (*p == '"') {
        ++p;
    }
    i = 0;
    if (dst_len > 0) {
        --dst_len;
    }
    while (*p) {
        if (*p == '"' && (p[1] == ',' || p[1] == '\r' || p[1] == '\n') || 
        (*p == '\r' || *p == '\n')) {
            ++p;
            break;
        }
        if (dst != NULL) {
            if (i < dst_len) {
                *dst++ = *p;
                ++i;
            } else if (!trim) {
                break;
            }
        }
        ++p;
    }
    if (dst != NULL) {
        *dst = 0;
    }
    *src = p;
    return 1;
}`

MaJerle, 2020-01-02:
This function is meant to process proper strings, including `"` at the end, before reaching CRLF characters.

wishmiss, 2020-01-03:
But if the string dose not end with **"**, such as model serial number command. After parsing, this string contains CRLF.

closed 2020-10-06 by wishmiss