MaJerle/lwcell #23: AT Messages Debug

yldzmuhammed, 2020-01-02:
Hello,

Is it possible to see sended and received at messages between mcu and module?

I enabled all the debug modes but no luck.

I dont want to make any changes on the library functions. But i think it could add on the received data process function. Am i correct?

MaJerle, 2020-01-02:
Library does not support this feature as not all commands are processed through same internal channel. 
if this is necessary, low-level driver shall implement this debug.

yldzmuhammed, 2020-01-02:
Hi @MaJerle 

In fact, it would be perfect if it could be possible.

Can you guide me how should i do this?

Using STM32F7 with keil and freertos (with CMSIS V2).

MaJerle, 2020-01-02:
Well, there is a function `gsm_input_process` which accepts received characters for processing. You can call your *debugger rx* function afterwards with same data to send these to your PC (or wherever you want).