MaJerle/lwcell #25: Event Callback Config

yldzmuhammed, 2020-01-03:
Hello @MaJerle 

Great thanks again for this great job.

I implemented this library successfully for Quectel M66 and make it work for HTTP GET, POST requests via tcp connection.

But it was little hard to get received data for me. I thought there is a event callback function for this. I still do. But could not find it.

Can you help me with this event callback mechanism? How to use these and how to configure it?

MaJerle, 2020-01-03:
Check client.c example in snippets folder.
Please note that M66 must respond with data same way as SIM800, otherwise it wont be parsed.

yldzmuhammed, 2020-01-04:
Hello @MaJerle 

There is no such file in the snippets folder. But there is a netconn_client.c file in that folder. But no callback function in it. Do i miss something here?

MaJerle, 2020-01-04:
https://github.com/MaJerle/gsm-at-lib/blob/develop/snippets/client.c

yldzmuhammed, 2020-01-04:
Have you noticed that this file in develop tag. Not in master.

I am checking it.

MaJerle, 2020-01-04:
Develop branch is where I put new code. Master branch contains latest releases only.

yldzmuhammed, 2020-01-11:
Hello again @MaJerle 

How can i get conn evetns? should register for this events?

MaJerle, 2020-01-12:
Did you check file I have posted link to?
Keep in mind that you have different GSM modem vs SIM800/900 which has different AT response thus may not send you an event.

closed 2020-08-05 by MaJerle