MaJerle/lwcell #33: Having issue using the  gsm_at_lib with STM32F4 discovery board on STM32CUBEIDE

smartraysam, 2020-06-09:
Hello @MaJerle

Great thanks for this great job.
please, I have tried to use the library with the STM32F4dicovery board on STM32CUBEIDE but I can't get it to compile.
 and also am little confused about this instruction line
1. Add port architecture gsm_at_lib/src/include/system/port/_arch_ folder to include path of your toolchain.. 
 what does this _arch_ mean because I can't see the folder in the path

2. Add source files from gsm_at_lib/src/ folder to toolchain build

Add source files from gsm_at_lib/src/system/ folder to toolchain build for arch port.

I will be expecting your reply thanks

MaJerle, 2020-06-23:
_arch_ is template name only. There are several architectures listed in the repository, like STM32 or WIN32. Inside each are include files that are necessary for the application to run.

smartraysam, 2020-06-25:
got it thanks

closed 2020-06-25 by smartraysam