MaJerle/lwcell #36: There is no interface layer with other third party library

keyhooon, 2020-08-05:
Hello @MaJerle

This is very nice and great code.

I am using your library successfully for sim808 and work with mqtt app.

i am confused how i can use from other library on top of your code?, how can use lwip or new library that ST released which name is Network library or Cloud general library on top of your code?

do you have plan for implement standard interface that we can use other library on top of your code?

for example if you implement interface like netif in lwip, your library can work with other third party libraries.

MaJerle, 2020-08-05:
What do you believe is standard interface?

keyhooon, 2020-08-06:
For example,Release Notes for
STM32 Network Library Says : "The STM32 Network Library is a middleware providing network services on STM32 devices. It provides a socket API (BSD like style) with support of secure or non secure connection and an API to control the lifecycle of the network adapters.

Three classes of network adapters are supported WIFI, Ethernet and Cellular. Different WIFI modules are supported from third party vendors."
This library provide standard Socket API (BSD like style), then other library that based on socket API can use it and provide feature on top of this socket API.
This library (Network Library) needs netif layer files to recognize type of network.
i'm not good in English, sry for that.

keyhooon, 2020-08-12:
For example i need TLS on MQTT, if you prepare socket layer like BSD socket, i can implement TLS on that and then Mqtt base on TLS socket.

MaJerle, 2020-08-12:
Most likely you need TLS on transport layer itself, so on TCP. This means starting SSL connection from modem point of view (that has full TCPIP stack).

closed 2021-03-30 by MaJerle