MaJerle/lwcell #38: Network connection problem with SIM7000E

ksuprynowicz, 2020-09-25:
Hello!
I'm having a problem with LwGSM on SIM7000E. I ported the library to ESP32+FreeRTOS, and sending SMS works fine, but I can't get connected to MQTT server. SIM7000E module is reset to factory settings, excluding baud rate (LwGSM didn't seem to work with auto-baud).
lwgsm_init fails with lwgsmTIMEOUT. Later LwGSM tries AT+CNUM a few times but modem replies with ERROR.
During the lwgsm_network_attach, LwGSM sends AT+CGACT=0, replied with OK, and AT+CGACT=1 replied with ERROR.
Later when library tries to connect to the server, it sends AT+CIPSSL=0, which command is not supported by the SIM7000, it only works on SIM800/900, so modem replies with ERROR, and function returns LWGSM_MQTT_CONN_STATUS_TCP_FAILED.
I'm seriously stuck on this one. Any help would be greatly appreciated.

MaJerle, 2020-09-27:
SIM7000E is unfortunately not directly supported with this library. It does not have the same AT commands set and would need (almost) complete re-write of low-level library part.

ksuprynowicz, 2020-09-27:
Thank you for the reply!
I was confused abouth this, because in features it says that SIM7000 modules are supported, and it mostly works well - I was able to succesfully send SMS and unencrypted MQTT messages, but I haven't checked rest of the library (I only need encrypted MQTT for my project).
Is there a way to send custom AT commands from LwGSM to modem?

nickfaughey, 2020-10-30:
> SIM7000E is unfortunately not directly supported with this library. It does not have the same AT commands set and would need (almost) complete re-write of low-level library part.

This is indeed a confusing statement; the README states that the library is compatible with SIM7000 (SIM7000E is just a band variant), and the AT command library is nearly identical, save for some LTE-specific features. Is there any reason SIM7000's shouldn't be expected to work with lwgsm?

ksuprynowicz, 2020-10-30:
It mostly works, but there are some problems. I spent a lot of time on porting it to ESP32 and thinking that there was problem in my implementation, because README said that SIM7000 is supported.

nickfaughey, 2020-12-01:
FWIW I have this working seamlessly on a SIM7000A right now, as long as `LWGSM_CFG_NETWORK_IGNORE_CGACT_RESULT` is set to 1 as you mentioned originally. I assume there may be some divergence if you need SIM7000-specific functionality that's not implemented in the library, but the core commands are identical and work just fine for my SIM7000A.

fizzyade, 2023-01-06:
I have been trying to get this working with a SIM7000E, I have had intermittent connections that work, but out of 8 hours today I've managed to see it connect twice to the internet.

While going through the code I found the flag LWGSM_CFG_NETWORK_IGNORE_CGACT_RESULT and set that to 1, I didn't think It did anything but in the time between me setting that and having a conversation with a colleague I discovered it had connected and it had published to an MQTT server, since then it has not managed to do it.

At that point I found this issue and we do have a SIM7000E.

What's confusing is that there doesn't appear to be a "SIM7000" device, we cannot find any mentioned anywhere of a SIM7000 device without a suffix.

While this issue is specifically about the "7000E", @MaJerle states in this that only the "7000" is supported, a product that we cannot find any existence of?

Can you clarify the state of support for the SIM7000 in here because since seeing this issue we're now very confused as to what SIM7000 modules are actually supported.

We are also testing using a 1nce data sim.

Thanks

MaJerle, 2023-01-06:
Support for 7000 (maybe w/ E suffix, don't recall anymore) has been marginally tested. It was more a review of the (some of) at commands. 

There is an extensive work to be done - actually long-term plan (for which I need find time) is to rework library to rather work on "services". We can then split the drivers for different low-level modem, where each of them has to implement specific service.