MaJerle/lwcell #40: lwgsm-0.6.0 release

gadak, 2020-10-12:
gsm_sys.h
line 78 esp instead of gsm

are there any patches to correct these?

I first tested on lwgsm-develop branch and lwgsm_init(lwgsm_callback_func, 1) never returned. all commands response was being received proper on a STM32F103 Board I have.

Then switched to release branch to find that naming convention of files was different.

Could you please explain a bit.

Thanks for the library!!!

MaJerle, 2020-10-12:
0.6.0 is as is, no patches available.

Where does lwgsm fails to return? Where does it stuck when entering debug mode?

gadak, 2020-10-12:
In lwgsm-develop branch - 

Function trace - 

Within - lwgsm_init(lwgsm_callback_func, 1); called from init_thread - 

@ line no - 141 of (lwgsm.c) res = lwgsm_reset_with_delay(LWGSM_CFG_RESET_DELAY_DEFAULT, NULL, NULL, blocking);  /* Send reset sequence with delay */

@ line no 2306 of (lwgsm_int.c)

    if (res == lwgsmOK && msg->is_blocking) {   /* In case we have blocking request */
        uint32_t time;
        time = lwgsm_sys_sem_wait(&msg->sem, 0);/* Wait forever for semaphore */
        if (time == LWGSM_SYS_TIMEOUT) {        /* If semaphore was not accessed within given time */
            res = lwgsmTIMEOUT;                 /* Semaphore not released in time */
        } else {
            res = msg->res;                     /* Set response status from message response */
        }

-----------------
How to print debug log? Do we need to redirect it to printf? As I am not able to verify what happens in the background

MaJerle, 2020-10-14:
By default, if you enable debug feature, it will use `printf` to redirect stream.

https://docs.majerle.eu/projects/lwgsm/en/latest/api-reference/lwgsm/debug.html

PouyaMalek, 2020-11-30:
> In lwgsm-develop branch -
>
> Function trace -
>
> Within - lwgsm_init(lwgsm_callback_func, 1); called from init_thread -
>
> @ line no - 141 of (lwgsm.c) res = [...]

Hello,I have same problem too. on stm32f103 , i debug my program and i found that interrupt routins of serial doesnt work

closed 2021-03-28 by MaJerle