MaJerle/lwcell #43: PWRKEY pin

blomnik, 2020-11-12:
Hello!
I have a custom PCB with PWRKEY pin instead of RST. Is it possible utilize it on init with lwGSM or I have to implement some additional code by myself?

Thank you.

nickfaughey, 2020-12-01:
You can implement whatever custom bootup logic you want in your [`lwgsm_ll_init()` function](https://github.com/MaJerle/lwgsm/blob/develop/lwgsm/src/system/lwgsm_ll_template.c#L68). For example, this is where I wake up my SIM7000 module from sleep mode by pulling its DTR pin low. If you have a board that requires some use of the PWRKEY pin like pulsing it high or low for some time, you could do it here.

blomnik, 2021-01-24:
So, my workaround is
```c

/**
 * This function try to init GSM modem and power it on by  PWRKEY if init was failed
 */
int gsm_reset(void) {
  for (int i = 0; i < 3; ++i) {
    if(lwgsm_init(NULL, 1) != lwgsmOK) {
      int err;
      if(GSM_PWRKEY_port) {
        LL_GPIO_ResetOutputPin(GSM_PWRKEY_port, GSM_PWRKEY_pin);
        osDelay(1500); // according to 4.2.2.1 of "SIM800 hardware design"
        LL_GPIO_SetOutputPin(GSM_PWRKEY_port, GSM_PWRKEY_pin);
        osDelay(1500);

        err = lwgsm_reset_with_delay(LWGSM_CFG_RESET_DELAY_DEFAULT, NULL, NULL, 1);
        if(err != lwgsmOK) {
          printf("lwGSM error %d", err);
        }
      }
      continue;
    }

    return 0;
  }

  errno = ENODEV;
  return -1;
}
```