MaJerle/lwcell #44: Guidance for implementing LWGSM_CFG_CONN_MANUAL_TCP_RECEIVE

nickfaughey, 2020-12-01:
I'm facing an issue with the automatic TCP data retrieval (`CIPRXGET=0`), where my socket layer's receiving ring buffer is getting overrun by GSM data coming in faster than my application can read and process it. The solution seems to be to switch the GSM architecture to use `CIPRXGET=1` manual data retrieval so that my app can control the ingestion rate. My initial thought was to implement the AT commands for manually receiving TCP data (`AT+CIPRXGET=1,n,len`) and directly call it from my socket layer, but I stumbled upon [this lwgsm_conn_recved() stub method](https://github.com/MaJerle/lwgsm/blob/develop/lwgsm/src/lwgsm/lwgsm_conn.c#L323), which looks like it was intended to be implemented for this purpose. What was the original intention and roadmap for this function, and does it make sense to piggyback off of the stub started there instead of manually invoking and reading the `CIPRXGET` data response?

MaJerle, 2020-12-01:
Idea of recved method is to notify stack that application used data and does not need them anymore, or better to say, that application has now more resources to process more data. Imagine if there would be slow implementation of file system and data stream would be faster than processing.

I have in fact another idea in mind (it was implemented in lwesp at least) that when in manual TCP receive, first thing I do is to try to allocate memory for buffer, if successful, then I try to read data from device, if not, then I have to try on next run. To avoid constant trying, one of the ideas was to *signal* memory ready from recved function.

For compatibility reasons, keep a call to recved function as in examples.

nickfaughey, 2020-12-01:
Ah, I had no idea this exact scenario was implemented in the lwgsm library, I'll mirror my lwgsm changes off of that library's manual data receive architecture. Looks like the same concept should apply to SIM800/7000 as ESP.

closed 2020-12-01 by nickfaughey