MaJerle/lwcell #45: How difficult to change this middleware for Quectel modem?

xiongyu0523, 2020-12-02:
I understand this lib is intended to be used with SIMCOM modem. Is it easy to port to use with a different modem like quectel?

MaJerle, 2020-12-02:
I was hoping that I could make it easily to port, but in reality this means change of all AT command calls, parsing of all responses. Only high-level API calls would be the same. In reality, work load would be almost the same as keeping and maintaining 2 libraries.

Almost all `lwgsm_int.c` file would need to be rewritten + header files for commands.

yldzmuhammed, 2020-12-03:
I actually succesfully worked with M66 modem with changing this library. I can fork it if you like.

xiongyu0523, 2020-12-04:
Thanks all. @MaJerle That sounds a lot work to do.  But I guess there is no structure level thing need to be changed. right?
@yldzmuhammed, yes please, could you make it open source?