MaJerle/lwcell #49: AT+CIPSTART wrong format for SSL

gxnt-samir, 2021-03-16:
Hello @MaJerle ,

I was trying to enable the SSL and it seems there is some issue with the AT cmd for SSL.

This is the snipet from my actual log files using lastest release source.

AT+CIPSSL=1
AT+CIPSSL=1
OK
AT+CIPSTART=5,"device.connectivity-suite.cloud",443
AT+CIPSTART=5,"device.connectivity-suite.cloud",443
+CME ERROR: 3

According to the AT cmd document from SIMCOM for SIM800 series command format is as follows.

AT+CIPSSL=1  
OK
AT+CIPSTART=1,"TCP","116.228.221.51","9600"  
OK

1,CONNECT OK

I think you are missing "TCP" in your format and dobule quote in the port 

Can you please let me know how can i patch these in my code. I am actually stuck in the middle of my workflow due to this.

I am looking forward for your reply.

Thank and Regards,
Samir

gxnt-samir, 2021-03-16:
Hello @MaJerle,

Just for your reference, to do a quick fix i modified the code as follows in lwgsm_int.c Line no 1948 as follows

            AT_PORT_SEND_CONST_STR("+CIPSTART=");
            lwgsmi_send_number(LWGSM_U32(c->num), 0, 0);
            if (msg->msg.conn_start.type == LWGSM_CONN_TYPE_TCP) {
                lwgsmi_send_string("TCP", 0, 1, 1);
                lwgsmi_send_string(msg->msg.conn_start.host, 0, 1, 1);
                lwgsmi_send_port(msg->msg.conn_start.port, 0, 1);
            } else if (msg->msg.conn_start.type == LWGSM_CONN_TYPE_UDP) {
                lwgsmi_send_string("UDP", 0, 1, 1);
                lwgsmi_send_string(msg->msg.conn_start.host, 0, 1, 1);
                lwgsmi_send_port(msg->msg.conn_start.port, 0, 1);
            } else{
                lwgsmi_send_string("TCP", 0, 1, 1);
                lwgsmi_send_string(msg->msg.conn_start.host, 0, 1, 1);
                lwgsmi_send_port(msg->msg.conn_start.port, 1, 1);
            }
            AT_PORT_SEND_END_AT();

It simply add the changes when the SSL is set and remain same duting others.

Please let me know your feedback.

Thanks And Regards,
Samir

MaJerle, 2021-03-17:
Yes this is because SSL was not considered

closed 2021-03-17 by MaJerle