MaJerle/lwcell #50: Support need for TCP : +RECEIVE missed event

gxnt-samir, 2021-03-18:
Hello @MaJerle,

I have send a request on TCP and expecting server to respond.

The server respond too but missed the first reveive to call the function callback.
I have marked the breakpoint in the attached log.

For simulation when i send dummy data from server the 2nd time, it hit the marked breakpoint.
I have tested manytimes and tried many approach. But it always miss the 1st response from the server.

Can you please support. In my real application the server is actually responding once. if missed the important data is lost.
Can you suggest any alternatives to track this receive buffer by application?

Regards,
Samir

[image]

MaJerle, 2021-03-18:
Hi @gxnt-samir more info would be necessary. We need to see 2 things:

- Actual data from modem
- Data visible by your microcontroller or computer. Are you sure that you are not missing any bytes received from the modem?

gxnt-samir, 2021-03-18:
Hello @MaJerle ,

Thank you for your response.

I am attaching the actual MCU - Modem communication log.
[image]

gxnt-samir, 2021-03-18:
As you can see, the +RECEIVE has been received from Modem interface with the response data, but this received data didnot triggered the event.

MaJerle, 2021-03-18:
Question is rather, did MCU really got these data from the modem? Or was there read data processing error?

gxnt-samir, 2021-03-18:
This print is executed by the MCU at just before passing to lwgsm_input_process API. You can see my iostream write API just before it.

[image]

gxnt-samir, 2021-03-18:
So, Yes the data is actually receiced by MCU, and its 1460 bytes. The image is croped so the data is not showing full. Do you have any way i can debug further. any specific function or value that can give me any clue.

MaJerle, 2021-03-18:
It is super strange. Would be good to see if you go in to input_process function.

closed 2023-07-29 by MaJerle