MaJerle/lwcell #52: What are the hardware requirements

adam-esp, 2021-03-29:
RAM 
ROM
I/O
?

MaJerle, 2021-03-29:
Based on arch and your system implementation.

adam-esp, 2021-03-29:
I understand but how much ROM do I need for the LWgsm library?

MaJerle, 2021-03-29:
This is not enough info. What is the arch you are using and what configuration would it be?

Using full configuration with GCC v9 for ARM, it took `21.34kB` of ROM memory. With full optimization (for size), it took `13kB`. This includes only LwGSM part and not low-level functions that are system important.

adam-esp, 2021-03-29:
STM32Fxxx will only handle lwgsm support and communicate with ESP32 over SPI. I have too little I / O on ESP32, that's why this STM32.
Do you think that 64kB ROM and 64kB RAM should be enough?

MaJerle, 2021-03-29:
Seems OK for me.

closed 2021-03-29 by MaJerle