MaJerle/lwcell #53: Native MQTT support

d3vil-st, 2021-03-30:
Hi there,
is there any plans to implement native MQTT connections for modules like sim7000 or sim800?
Here is doc for this: https://simcom.ee/documents/SIM7000x/SIM7000%20Series_MQTT_Application%20Note_V1.00.pdf

MaJerle, 2021-03-31:
Native MQTT support is only for SIM7000. Challenging for me at the moment to implement this.

If you are willing to make it happen, I have no prob to merge your work.

d3vil-st, 2021-03-31:
I’ve already started implementation for sim800 (sim7000 very similar, but have some differences during configuration), all basic commands work well (conf, connect, subscribe, publish and parse unsolicited responses). Will share create PR once done. I’ve asked just to sure that I’m not doing that perhaps already implemented.

d3vil-st, 2021-04-05:
#55 here is implementation of basic functions.