MaJerle/lwcell #54: Hardfault error sometimes

vinhemotiv, 2021-04-03:
Hello,
Sometimes, I get hardfault error when calling this function to attach network? I'm using Sim800C with STM32
 while (lwgsm_network_request_attach() != lwgsmOK) {
        lwgsm_delay(1000);
				printf("Network request attach FAILED \r\n");
			 	if(count_connect++ >5) 
					 {
						 goto terminate;
					 }
    }
Any idea to resolve it?

MaJerle, 2021-04-03:
Doyou know what was instruction causing hardfault? Which core do you use?

vinhemotiv, 2021-04-03:
I'm trying to debug it. Is it a blocking function? will loop forever if it causes any error? I'm using mqtt client example.
I did use your old version cmsis-rtos v1, and it looks more stable than v2? Why did you change it to v2?

vinhemotiv, 2021-04-03:
Is it possible to turn off GSM module after connection? I want to send data 10 minutes every times but don't want to keep it on to save battery

MaJerle, 2021-04-03:
Cmsis-os2 is working well. Function is blocking yes.

vinhemotiv, 2021-04-09:
I'm trying to use Sim7000E. I have to send a few commands "AT" to get "OK" to config the auto baurt rate before initializing the library. Is there anywhere in the lib to do that? I recognize that it won't work if Simcom has not been configured the COM rate automatically.

MaJerle, 2021-04-09:
This implementation is not available in the lib at the moment