MaJerle/lwcell #61: I am not getting any response with sim7600e

sushant-ht, 2021-07-21:
i am trying to use lwgsm library with esp32 and sim7600e, and i am facing problem, i get 

> Cannot initialize LwGSM

i have changed following steps in Library

/* Defines ESP uart number to use */
#define GSM_UART_NUM UART_NUM_1

ESP_ERROR_CHECK(uart_set_pin(GSM_UART_NUM, 26, 25, 0, 0));

i had written an uart routine to check AT command with same UART setting i do get AT response from modem.
i have insert a debug line in uart events routine, this line is never invoked
```
case UART_DATA:
			**printf("[UART DATA]: %d", event.size);**
          uart_get_buffered_data_len(GSM_UART_NUM, &buffer_len);
```

MaJerle, 2021-07-21:
Code you are showing is not library specific.

sushant-ht, 2021-07-21:
i modified lwgsm_ll_esp32.c 

`/**
 * \file            lwgsm_ll_esp32.c
 * \brief           Low-level communication with GSM device for Esp32
 */

/*
 * Copyright (c) 2021 Tilen MAJERLE and Ilya Kargapolov
 *
 * Permission is hereby granted, free of charge, to any person
 * obtaining a copy of this software and associated documentation
 * files (the "Software"), to deal in the Software without restriction,
 * including without limitation the rights to use, copy, modify, merge,
 * publish, distribute, sublicense, and/or sell copies of the Software,
 * and to permit persons to whom the Software is furnished to do so,
 * subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be
 * included in all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND,
 * EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES
 * OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE
 * AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT
 * HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY,
 * WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
 * FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR
 * OTHER DEALINGS IN THE SOFTWARE.
 *
 * This file is part of LwGSM - Lightweight GSM-AT library.
 *
 * Authors:          Tilen MAJERLE <tilen@majerle.eu>,
 *                   Ilya Kargapolov <d3vil.st@gmail.com>
 * Version:         v0.1.0
 */
#include "system/lwgsm_ll.h"
#include "lwgsm/lwgsm.h"
#include "lwgsm/lwgsm_mem.h"
#include "lwgsm/lwgsm_input.h"
#include "driver/uart.h"
#include "esp_log.h"

#if !__DOXYGEN__

/* Define TAG for log messages */
#define TAG "lwGSM"

/* Defines ESP uart number to use */
#define GSM_UART_NUM UART_NUM_1

#if !defined(LWGSM_USART_DMA_RX_BUFF_SIZE)
#define LWGSM_USART_DMA_RX_BUFF_SIZE      0x1000
#endif /* !defined(LWGSM_USART_DMA_RX_BUFF_SIZE) */

#if !defined(LWGSM_MEM_SIZE)
#define LWGSM_MEM_SIZE                    0x1000
#endif /* !defined(LWGSM_MEM_SIZE) */

static QueueHandle_t gsm_uart_queue;

static uint8_t initialized = 0;

char* uart_buffer[LWGSM_USART_DMA_RX_BUFF_SIZE];

/**
 * \brief           Send data to GSM device, function called from GSM stack when we have data to send
 * \param[in]       data: Pointer to data to send
 * \param[in]       len: Number of bytes to send
 * \return          Number of bytes sent
[... 131 lines cut ...]
        ll->send_fn = send_data;                /* Set callback function to send data */
    }

    /* Step 3: Configure AT port to be able to send/receive data to/from GSM device */
    configure_uart(ll->uart.baudrate);          /* Initialize UART for communication */
    xTaskCreate(uart_event_task, "uart_lwgsm_task0", 4096, NULL, 5, NULL);
    initialized = 1;
    return lwgsmOK;
}

/**
 * \brief           Callback function to de-init low-level communication part
 * \param[in,out]   ll: Pointer to \ref lwgsm_ll_t structure to fill data for communication functions
 * \return          \ref lwgsmOK on success, member of \ref lwgsmr_t enumeration otherwise
 */
lwgsmr_t
lwgsm_ll_deinit(lwgsm_ll_t* ll) {

    ESP_ERROR_CHECK(uart_driver_delete(GSM_UART_NUM));
    initialized = 0;                            /* Clear initialized flag */
    return lwgsmOK;
}

#endif /* !__DOXYGEN__ */
`

sushant-ht, 2021-07-21:
i have modified UART_event example from esp32 idf and it works well, i am atleast getting AT response
here code and log snippet

`    /* Configure parameters of an UART driver,
     * communication pins and install the driver */
    uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE
    };
    uart_param_config(EX_UART_NUM, &uart_config);

    //Set UART log level
    esp_log_level_set(TAG, ESP_LOG_INFO);
    //Set UART pins (using UART0 default pins ie no changes.)
	//-------------------------tx--rx
    uart_set_pin(EX_UART_NUM, 26, 25, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
   //uart_set_pin(EX_UART_NUM, 25, 26, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
    //Install UART driver, and get the queue.
    uart_driver_install(EX_UART_NUM, BUF_SIZE * 2, BUF_SIZE * 2, 20, &uart0_queue, 0);`

`[0;32mI (15771) scan: [Command]: AT
[0m
[0;32mI (15771) scan: uart[1] event:[0m
[0;32mI (15771) scan: [UART DATA]: 9[0m
[0;32mI (15771) scan: [IN DATA] AT

OK

[0m`

sushant-ht, 2021-07-22:
It was solved It was my mistake as RESET line stayed LOW.

closed 2021-07-22 by sushant-ht