MaJerle/lwcell #62: Error in FreeRTOS port

elAlexOmDiWo, 2021-07-22:
I want to express my gratitude for your work and point out ( unfortunately ) an error -
https://github.com/MaJerle/lwgsm/blob/433e80a49f1951c25466ab3e7f18b7c3471f5df5/lwgsm/src/system/lwgsm_sys_freeRTOS.c#L51-L56

sys_mutex is created as recursive and try give ( ? ) as regular.

And unprotect ( L91-L95 ) correctly like this - 

uint8_t
lwgsm_sys_unprotect(void) {
    return xSemaphoreGiveRecursive(sys_mutex);
}
https://www.freertos.org/a00123.html
<This macro must also not be used on semaphores created using xSemaphoreCreateRecursiveMutex().>

MaJerle, 2021-07-24:
Thanks, I have updated the code.

closed 2021-07-24 by MaJerle

elAlexOmDiWo, 2021-07-24:
Thank you for your work.