MaJerle/lwcell #63: how many files and how much work it will need to adapt library to support SIM7600

sushant-ht, 2021-07-23:
Hi, Tilen Majerle

Nice work really liked your work, I just want to know how much change i have to make to adapt this library to support sim7600. i believe lwgsm_input.c is the file that has the main send and receives AT command parsing. just to know is there any other file that needs to be changed? 

Thanks for Library and support
Sushant

MaJerle, 2021-07-24:
Most work is to be done in lwgsm_int and lwgsm_parser files

sushant-ht, 2021-09-08:
Why i am unable to read SMS, i am trying to debug this library with lpc1768 & sim800c,  but i am getting problems to read,

i have added log below

> Starting GSM application!
> Library initialized!
> < 
> +CMTI: "ME",42
> ,18> AT,2> +CFUN=1,1,9> 
> ,2< AT+CFUN=1,1
> 
> OK
> ,18> AT,2> E1,2> 
> ,2< ATE1
> 
> OK
> ,11> AT,2> +CFUN=,6> 1,1> 
> ,2< AT+CFUN=1
> 
> OK
> ,16> AT,2> +CMEE=1,7> 
> ,2< AT+CMEE=1
> 
> OK
> ,16> AT,2> +CGMI,5> 
> ,2< AT+CGMI
> 
> SIMCOM_Ltd
> 
> OK
> ,28> AT,2> +CGMM,5> 
> ,2< AT+CGMM
> 
> SIMCOM_SIM800C
> 
> OK
> ,32> AT,2> +CGSN,5> 
> ,2< AT+CGSN
> 
> 867157040500092
> 
> OK
> ,33> AT,2> +CGMR,5> 
> ,2< AT+CGMR
> 
> Revision:1418B06SIM800C24
> 
> OK
> ,43> AT,2> +CREG=1,7> 
> ,2< AT+CREG=1
> 
> OK
> ,16> AT,2> +CLCC=1,7> 
> ,2< AT+CLCC=1
> 
> OK
> ,16> AT,2> +CPIN?,6> 
> ,2< AT+CPIN?
> 
> +CPIN: READY
> 
> OK
> ,31> AT,2> +CNUM,5> 
> ,2< 
> Call Ready
> 
> AT+CNUM
> 
> OK
> ,30> AT,2> +CPMS=?,7> 
> ,2< AT+CPMS=?
> 
> +CPMS: ("SM","ME","SM_P","ME_P","MT"),("SM","ME","SM_P","ME_P","MT"),("SM","ME","SM_P","ME_P","MT")
> 
> OK
> ,119> AT,2> +CPMS?,6> 
> ,2< AT+CPMS?
> 
> +CPMS: "ME",22,50,"SM_P",20,20,"SM_P",20,20
> 
> OK
> ,62SMS enabled. Send new SMS from your phone to device.
> Start by sending first SMS to device...
> < 
> +CMTI: "ME",43
> ,18New SMS received!
> Cannot read SMS!

sushant-ht, 2021-09-08:
Why i am unable to read SMS, i am trying to debug this library with lpc1768 & sim800c,  but i am getting problems to read,

i have added log below

> Starting GSM application!
> Library initialized!
> <
> +CMTI: "ME",42
> ,18> AT,2> +CFUN=1,1,9>
> ,2< AT+CFUN=1,1
>
> OK
> ,18> AT,2> E1,2>
> ,2< ATE1
>
> OK
> ,11> AT,2> +CFUN=,6> 1,1>
> ,2< AT+CFUN=1
>
> OK
> ,16> AT,2> [...]

sushant-ht, 2021-09-08:
Why I am unable to read SMS, I am trying to debug this library with lpc1768 & sim800c,  but I am getting problems reading,

I have added the log below

> Starting GSM application!
> Library initialized!
> <
> +CMTI: "ME",42
> ,18> AT,2> +CFUN=1,1,9>
> ,2< AT+CFUN=1,1
>
> OK
> ,18> AT,2> E1,2>
> ,2< ATE1
>
> OK
> ,11> AT,2> +CFUN=,6> 1,1>
> ,2< AT+CFUN=1
>
> OK
> ,16> AT,2> [...]

[Uploading esp32.log…]()

sushant-ht, 2021-09-08:
This is a bug, in the below function you need to check for NULL condition,
else following snippet fails and it invoke fault handler
 /* Create SMS thread */
    _lwgsm_sys_thread_create(NULL, "lwgsm_sms", (lwgsm_sys_thread_fn)sms_send_receive_thread, NULL, 1024, LWGSM_SYS_THREAD_PRIO);_

uint8_t
lwgsm_sys_thread_create(lwgsm_sys_thread_t* t, const char* name, lwgsm_sys_thread_fn thread_func,
                      void* const arg, size_t stack_size, lwgsm_sys_thread_prio_t prio) {
    lwgsm_sys_thread_t id = NULL;
    if (xTaskCreate(thread_func,
                    name,
                    1024,
                    //stack_size/sizeof(portSTACK_TYPE)*2,
                    arg,
                    prio,
                    id) == pdPASS) {
    	**if(t != NULL)**
      *t = id;
      return 1;
    }
    return 0;
}

sushant-ht, 2021-09-09:
i have currently disabled CHECK_READY(); in fun lwgsm_sms_send & lwgsm_sms_read, i dont always receive SMS Ready from sim800c.

MaJerle, 2021-09-09:
You should not remove check ready step. How your send SMS API looks like? Complete one please.

Did you check this one? https://github.com/MaJerle/lwgsm/blob/develop/snippets/sms_send_receive.c