MaJerle/lwcell #65: MQTT not send data

aminmokhtari94, 2021-09-23:
Hi Mr. Tilen,
I'm using lwgsm on esp8266_rtos_sdk & sim868
the problem is on mqtt, it's connect to the server but hangs on Poll and does not send any data!
Best Regards

[image]

MaJerle, 2021-09-23:
I can see you print every received character on console, which is very good. I don't see that you actually send ">" character to stack.

App waits "> " 2 characters before AT+CIPSEND accepts data.

Could you also add a print info of data being sent from MCU to device?

aminmokhtari94, 2021-09-23:
Green lines are data that sent from MCU to Sim and whites are received.
Sim waiting for data by sending > but MCU don't do anything!
Am I right?

MaJerle, 2021-09-23:
Are you sure that ">" is going to process function? Can you debug whats going on?

aminmokhtari94, 2021-09-25:
Unfortunately i cant debug whats going on but yes it's work with netconn_client and sent http get request but for mqtt it's not.

[image]

aminmokhtari94, 2021-09-29:
The problem was the my low level send_data function, solved!

closed 2021-09-29 by aminmokhtari94