MaJerle/lwcell #69: Plans to release PPP support?

Spider84, 2022-03-18:
Hello!
Do you plan to release PPP support for next process f.e. by lwip pppos?

May be I can help?
Now I discovered for structure of library and have many questions - how to make better.

MaJerle, 2022-03-18:
I am listening :)

Spider84, 2022-03-18:
I has take IPCONN as example. 
```c
lwgsmr_t
lwgsm_ppp_start(void* const arg, lwgsm_evt_fn conn_evt_fn, const uint32_t blocking) {
    LWGSM_ASSERT("conn_evt_fn != NULL", conn_evt_fn != NULL);

    return lwgsm_call_start("*99#", arg, conn_evt_fn, blocking);
}
```
in `lwgsm_init.c` has added section in parser
```c
#if LWGSM_CFG_PPP
        } else if (CMD_IS_CUR(LWGSM_CMD_ATD)) {
                if (!strncmp(&rcv->data[0], "CONNECT" CRLF, 7 + CRLF_LEN)) {
			is_ok = 1;
                } else if (!strncmp(&rcv->data[0], "NO CARRIER" CRLF, 10 + CRLF_LEN)) {
			is_error = 1;
		}
#endif
```
now need to forward data from uart (`lwgsmi_process()`) to pppoe parser of lwip. 
How to do it better? Also need some event to detect end of data connection...

Spider84, 2022-03-21:
m? what kind?

MaJerle, 2022-03-21:
From your code

```
lwgsmr_t
lwgsm_ppp_start(void* const arg, lwgsm_evt_fn conn_evt_fn, const uint32_t blocking) {
    LWGSM_ASSERT("conn_evt_fn != NULL", conn_evt_fn != NULL);

    return lwgsm_call_start("*99#", arg, conn_evt_fn, blocking);
}
```

It seems that you are using `call` module of LwGSM library. The lib provides you events when call is being *picked up* or has been *hanged-up*

https://docs.majerle.eu/projects/lwgsm/en/latest/api-reference/lwgsm/evt.html#group___l_w_g_s_m___e_v_t_1LWGSM_EVT_CALL_CHANGED

closed 2023-07-29 by MaJerle