MaJerle/lwcell #83: There's something wrong with PHONEBOOK

edbek, 2023-12-13:
There's something wrong with PHONEBOOK.
When I try to change memory phonebook, for example, to ME, 
    lwcell_pb_list(LWCELL_MEM_ME, 1, pb_entries, LWCELL_ARRAYSIZE(pb_entries), &pb_entries_read, NULL, NULL, 1);
the command to change memory phonebook does not work.

When I set a debugging breakpoint at:
 } else if (CMD_IS_CUR(LWCELL_CMD_CPBS_SET) && !strncmp(rcv->data, "+CPBS", 5)) {
             lwcelli_parse_cpbs(rcv->data, 2); /* Parse +CPBS response */
 }
but there the program never stops on lwcelli_parse_cpbs(...).

MaJerle, 2023-12-13:
Can you tell me the return value of the lwcell_pb_list?

edbek, 2023-12-13:
lwcell_pb_entry_t       pb_entries[10];
size_t                          pb_entries_read;

        lwcell_pb_enable(NULL, NULL, 1);
        lwcell_pb_read(LWCELL_MEM_ME, 1, pb_entries, NULL, NULL, 1);
        lwcell_pb_list(LWCELL_MEM_ME, 1, pb_entries, LWCELL_ARRAYSIZE(pb_entries), &pb_entries_read, NULL, NULL, 1);

After executing this block of code, it turns out that pb_entries[0] and all the rest pb_entries[...] are equal:

mem = LWCELL_MEM_SM_P
pos = 0 
name = ""
number = ""
type = 0

MaJerle, 2023-12-13:
And return value of all 3 function calls?

edbek, 2023-12-13:
yes, after each of the 3 calls the data does not change and will remain so:
```

mem = LWCELL_MEM_SM_P
pos = 0
name = ""
число = ""
тип = 0

```
but I see that there is only one response from the modem during these 3 calls:
+CPBS: ("SM", "ME", "ON", "FD")

and I don’t see the АТ-command being called to install memory phonebook. The code never goes here:
```

case LWCELL_CMD_CPBS_SET: { /* Get current memory info */
              lwcell_mem_t mem;
             AT_PORT_SEND_BEGIN_AT();
             AT_PORT_SEND_CONST_STR("+CPBS=");

```

MaJerle, 2023-12-13:
Can you give me return values of these 3 functions please?

edbek, 2023-12-13:
all functions return lwcellOK

```
lwcellr_t ret;
       
        `ret = lwcell_pb_enable(NULL, NULL, 1);`
        here ret = lwcellOK

      `  ret = lwcell_pb_read(LWCELL_MEM_ME, 1, pb_entries, NULL, NULL, 1);`
        here ret = lwcellOK

       ` ret = lwcell_pb_list(LWCELL_MEM_ME, 1, pb_entries, LWCELL_ARRAYSIZE(pb_entries), &pb_entries_read, NULL, NULL, 1);`
       here ret = lwcellOK

MaJerle, 2023-12-13:
Can you enable full lib log and provide full at commands exchange between mcu and modem?

edbek, 2023-12-13:
Yes, I can, but it won't be soon. I just started working with the PYL modem and I don’t know what will happen.
I would like to get advice from you now.

Thank you.

MaJerle, 2023-12-13:
I am not able to give good advice at this point. I do not know the model you have nor I know if your driver properly works to enter all data.

It is strange that all functions return OK, too.