MaJerle/lwcell #85: Missing file !!!

danzadok, 2024-07-03:
Hi,  
1) I am trying to build LWCELL for windows using Cmake. It complains that a file is missing:
Cannot find source file:
   .../lwlib/lwcell/snippets/examples_common_lwcell_callback_func.c
2) Can you please let me know where do I change the compilation command to build for STM32?

Danny

MaJerle, 2024-07-03:
Demos for windows are not tested. I'm using only dev platform to test the points.
STM32 has been updated - you need to open example with cmake from inside stm32 folder and build with GCC.

Pull down latest commit, just uploaded.

danzadok, 2024-07-04:
The fix is on the main branch? I do not see it.
Do you recommend to work on the develop branch?

MaJerle, 2024-07-04:
It is on develop branch. Main is only when release is done.