MaJerle/lwcell #86: is the SSL functionality stable?

devprofile98, 2024-07-31:
Hi, Thank you for this great library, it solved many many problems for our team.

There is an option here for SSL connection over TCP at this [line](https://github.com/MaJerle/lwcell/blob/8ed628047c230f682db4d2fd460f6251e2f029b2/lwcell/src/include/lwcell/lwcell_netconn.h#L72), although I am generic error when using it, is this option stable? because I have seen you stated that SSL functionality on SIM devices is not that stable.

MaJerle, 2024-07-31:
Agree, the SIMCOM supports SSL connections, but i never managed to get it to work properly. SSL connections typically means that modem will perform handshake for you, also meaning that data on the UART will be in plain text.

Personally I see this as a problem and would prefer to make TCP/TLS handshake inside the MCU, but for that we need to establish plain TCP connection (not SSL) and then do the job on our own.

That being said, I had lack of time and motivation to continue with this implementation.

devprofile98, 2024-07-31:
I am experiencing the same problem. The SSL functionality on the SIM module works inconsistently and is not stable. It took me some time to realize that achieving stable SSL connections using only AT commands is not feasible.

Currently, I am porting MbedTLS to use the SIM868 as a transport layer and am handling the SIM868 using your library.

Would you mind if I tag this option as unstable in the source code?

MaJerle, 2024-07-31:
Feature is as stable as modem offers it. From AT commands, we cannot offer better :)

Sure, you can flag it, but for your specific modem only, please.

devprofile98, 2024-07-31:
yes, I understand that this is not a problem of LWCELL, SIM modem firmware is the one to blame.
Thanks for your response.

closed 2024-07-31 by MaJerle