MaJerle/lwcell #90: Sending Zero-Length Payloads to remove Retained MQTT Messages

devprofile98, 2024-09-14:
Hi! :hand: 

To remove a retained message from an MQTT broker, a subscriber must publish a zero-length payload to the broker on the same topic. However, in the `lwcell_mqtt_client_api_publish` function (as seen [here](https://github.com/MaJerle/lwcell/blob/590fb7787807ecdbf68983de2042012b0ca03bf9/lwcell/src/apps/mqtt/lwcell_mqtt_client_api.c#L406) and [here](https://github.com/MaJerle/lwcell/blob/590fb7787807ecdbf68983de2042012b0ca03bf9/lwcell/src/apps/mqtt/lwcell_mqtt_client_api.c#L407)), it is currently not possible to send a zero-length payload due to the use of assertions.

I understand the intent behind these assertions in the function, but I wanted to ask what the best approach would be to enable this functionality:

Should the assertions be removed to allow zero-length payloads?
Should the function's behavior be modified to specifically allow sending a zero-length message for retained messages?
Would it make sense to implement a new function to handle this use case?
Alternatively, should I rely on the underlying `lwcell_mqtt_client_publish` function for this purpose?
Looking forward to your thoughts on the best approach!

Thanks!

MaJerle, 2024-09-14:
I've removed the assert, giving the full control to the user to decide what to send.

It is allowed to send zero length payload, so it was a bug in the lib.

devprofile98, 2024-09-14:
Thank you so much!

closed 2024-09-14 by devprofile98